hpcreery/GRX #3: upload gerber file keeps loading forever

aeonSolutions, 2024-03-28:
Hi @hpcreery , @hpcreery 
I love your Gerber web renderer .  
awesome work. 

I'm trying to add it to all my projects here on Git Hub, however, when I upload a Gerber file the webpage does nothing. 
see the video below for a more detailed view of it.

https://github.com/hpcreery/GRX/assets/7936768/06506147-8226-4bb1-8454-4ea39078c259

can you tell me what am I doing wrong ?

hpcreery, 2024-03-28:
Hello @aeonSolutions ,
Thank you so much for the feedback!

From your video, you are doing everything correctly. What is likely happening is an unhandled exception during parsing the specific gerber file. If you can, please share your gerber file to allow me to debug the issue.

There is an active development happening in a fork here [gl-shaders-testing](https://github.com/hpcreery/GRX/tree/gl-shaders-testing) to rewrite the engine for the viewer. With the rewrite are many improvements are coming to the parser and renderer as well. Any found bugs from the provided feedback will be incorporated into the new version, if not already patched. I will be hoping to be able to merge the new version and make it public soon!

aeonSolutions, 2024-04-08:
cool Great.

I have another, unrelated question.

I'm trying to integrate your code into my website .
Is there the option to remove the Button "upload Gerbers" and instead load predefined gerber files . for instance from a configuration file ?

thank you

closed 2024-04-13 by hpcreery